ODZ-UJF-AV-CR/balon #13: Praskají bublinky? Vrzají esíčka?

vaclavstepan, 2015-12-16:
Máří včera nadnesla, že bublinky fakt popraskaj. Jestli jo, budem mít problém s tepelnou izolací a dopadem do tvrdého.

Koupil jsem preventivně na poště dva balíky esíček (je to něco jako lehký polystyren s uzavřenými póry). Možná to bude vrzat a rušit (zkusím testnout).

Bylo by fajn zkusit, jestli bublinky fakt praskaj nebo nepraskaj (v 30 km ekvivalentním tlaku).

vaclavstepan, 2015-12-16:
Dáša říká, že tlak v 30 km je cca 11 hPa.

kaklik, 2015-12-16:
Nevim jak takový tlak teď vyrobit. To už chce olejovou vývěvu..

kakl, 2015-12-16:
> Nevim jak takový tlak teď vyrobit. To už chce olejovou vývěvu.. Na FEL
> je na to sice simulátor, ale ten je nepoužitelnej.. Dostat se k němu
> je birokracie na dva dny. A tolik času nemáme.

Svetlik ma vyvevu, ale sklenena nadoba neni moc velka. Bublinky by 
vyzkouset sly.

vaclavstepan, 2015-12-16:
Jo, stačí pár bublinek a nebo ověřit, že esíčka nedělaj rušení.

vaclavstepan, 2016-06-01:
Esíčka byly shledány použitelnými. Bublinky byly shledány nepoužitelnými.

closed 2016-06-01 by vaclavstepan