ODZ-UJF-AV-CR/balon #25: Battery monitor

vaclavstepan, 2017-02-25:
Battery monitor reports nonsenses:

`BatTemp: -427.35 C RemCap: -1542 mAh FullCap: -1542 mAh U: -1542 mV I: -67078 mA Charge: -1542.00 %`

kaklik, 2017-02-25:
Za jakých podmínek se to stane?

vaclavstepan, 2017-02-25:
Kdykoliv. Pustim-li m_i2c.py i zkusím-li z pymlabu lion.py example (teda, verzi z minulyho roku).Vše včetně Temp se zdá konstantní a nesmyslné.Vasek                                                                                                                                                                                                                                                                                                                                        --Department of Radiation Dosimetry, Nuclear Physics Institute of the CAS, Prague, Czech Republic                                                                                                                                                                                                                From: Jakub KákonaSent: sobota, 25. února 2017 16:03To: ODZ-UJF-AV-CR/balonReply To: ODZ-UJF-AV-CR/balonCc: vaclavstepan; AuthorSubject: Re: [ODZ-UJF-AV-CR/balon] Battery monitor (#25)Za jakých podmínek se to stane?

vaclavstepan, 2017-02-25:
Martin Kákona vyřešil konektorem na I2C hubu, který zmenšuje možnost zapojit to blbě.

closed 2017-02-25 by vaclavstepan