ODZ-UJF-AV-CR/balon #33: Anténa k telemetrickému vysílači

kaklik, 2019-10-05:
Podobně jako v issue #30  telemetrický vysílač taktéž potřebuje anténu, avšak na jiné pásmo [433-434 MHz](https://www.ctu.cz/vyuzivani-vymezenych-radiovych-kmitoctu).

Anténa by taktéž měla být typu QFH a je potřebné, aby byla jak ve variantě pro balon tak i ve varinantě pro pozemní stanice (automobily s přijímací stanicí).

Pozemní varianta by se měla montovat na rovnou překližkovou desku a být odolná proti dešti, toto se předpokládám zajistí montáží kuželového vytištěného radomu přes celou anténu. 

Anténa může být do překližky uchycena jedním nebo více šrouby.

cernohorsky, 2019-10-07:
Pomalu se chystam na navrh tehle anteny, ale tuhle variantu bych docala rad udelal z koaxialniho kabelu a ne z postribrneho dratu. Koax by mnel vetsi prumer coz by mnelo zlepsit parametry anteny a muzu realizovat i "nekonecny balun", na druhou stranu antena bude tezsi. Ted ma verze pro 868 MHz 21.5g. Verze pro 433 MHz bude mit urcite vic jak 50 g, Jak moc jsme limitovani vahou celeho reseni?

cernohorsky, 2019-10-08:
Ta antena pro 433 Mhz je obrovska (v pozadi verze pro 868 Mhz):
[image]

A navyse porad neni spravne naladena, musim ji jeste vic zvetsit a uz tenhle model se mi sotva vesel do tiskarny:
[image]

[image]

kaklik, 2019-10-08:
Domnívám se, že průměr vodiče antény by na parametry antény, kromě šířky pásma mít vliv neměl, pokud bude zachována vodivost vodiče při různých průměrech. 

Postříbřeným drátem tak lze realizovat i tenké vodiče a anténa tak bude mít malou šířku pásma, což pokládám za výhodu. 

Nejde ten tištěný model nějak odlehčit?  Na celou gondolu máme předpokládám něco jako 1-2kg.

cernohorsky, 2019-10-08:
Taky se mi ta "koaxova varianta" moc nelibi. Nedokazu sni ani dodrzet spravne rozmery. O tom odlehceni jsem taky premyslel, ale rikam si jestli by nebylo nejlepsi,  obejit se uplne bez toho modelu. Kdyz jsem totiz delal variantu 868 MHz, tak jsem zkusil ten model pouzit jen jako kopyto a vysledek byl moc pekny:
[image]

Antena pak bude super lehka. Stacilo by pak zmerit jen vliv radomu.
Ma to jen dve uskali :
1.	Pro ground station toto reseni bude bezproblemove, ale jinak ta antena mechanicky nic nevydrzi. Staci jen lehky naraz a je po ni. Jeste by mozna stalo za to skusit ji udelat z klasickeho medeneho dratu. S prumerem 2mm by se mohlo porad dobre pracovat a antenna bude mnohem robustnejsi a presto porad lehka.
2.	Ten model bude vetsi nez 230mm a moje tiskarna uz tak velky model nevytiskne. Kopyto vytisknu z dvou kusu, ale radom by chtelo tisknout v celku a to uz nedam. Tiskne v Praze jeste nejaka RED-ka? 😉

kaklik, 2019-10-08:
Ta anténa by myslím mela vydržet pád gondoly na ní (anténa bude vespod) a je myslím hloupý po dopadu přijít o vysílání telemetrie.  Ten tištěný model by právě měl zajistit že se po dopadu anténa nezdeformuje. 

Myslím že je snažší to nakrájet na menší díly, než se snažit to tisknout vcelku. Radom může být klidně slepený a anténa taky.

kaklik, 2019-10-09:
Ještě mě k té hmotnosti napadlo, zda nejde hmotnost ušetřit na přívodním koaxu k té anténě. 
Ona totiž v podstatě stejně vyžaduje symetrický přívod, což koaxiální vedení nesplňuje. Také vidím potenciální problém s použitím SMA konektoru, protože je to typ konektoru který se povoluje a není to na něm poznat, máme s ním v tomhle smyslu špatné zkušenosti s použitím u GPS antén. 

Navrhoval bych vyzkoušet napájení antény [kroucenou dvoulinkou](https://www.rf-microwave.com/en/nbp/nmp/pvc-insulated-twisted-pair-silver-plated/frb-01/), jejíž připojení k asymetrickému výstupu měřáku lze realizovat tím [transformátorem ETC1-1-13](https://cz.mouser.com/datasheet/2/249/ETC1-1-13-335562.pdf) o kterém jsem už psal. 

Samotné připojení antén v balonu by bylo fajn, kdyby se obešlo bez RF konektoru. Ovšem všechny SiK modemy, které máme končí SMA, případně R-SMA konektorem.

cernohorsky, 2019-10-09:
Zmeril jsem hmotnost RG174 a 30 cm tohoto kabelu vazi jen 4,7 g. A pokud by tahle hmotnost presto vadila sel by pouzit pigtail. Ale ta kroucena dvoulinka neni uplne spatny napad, chtelo by to kouknout jak by to fungovalo. Co se tyce konektoru, tak nejjednoduchsi by bylo ty kabely natvrdo napajet na DPS, ale neni to moc prakticke reseni. Co takhle zkusit ten konektor pred letem zakapnout lakem na srouby?
https://eshop.wuerth-industrie.com/Kategorie-produktu/Lak-na-pojisteni-sroubu/15433005030601.cyid/1543.cgid/cs/CZ/EUR/

kaklik, 2019-10-09:
Jo, podobné lepidlo používáme pro GPS antény.  Nějak to funguje, jen se to musí udělat dobře, aby konektor zůstal vodivý. 
Ale fajn, v tom případě kabel z hlediska hmotnosti zřejmě není potřeba řešit.

kaklik, 2019-11-12:
Anténa vyrobená o víkendu, podle těchto parametrů: 

[image]

V takovémto provedení

[image]

Je naladěná takto: 

[image]

kaklik, 2019-11-28:
Model antény jsem iteračně upravil na správnou rezonanční frekvenci. 

[image]

[image]

Celá anténa včetně přívodu pak váží ~38g 

[image]

closed 2019-11-28 by kaklik