ODZ-UJF-AV-CR/balon #40: Zlepšit způsob připojení antén k balonu

kaklik, 2019-11-30:
Aktuální konstrukce antén pod gondolou funguje dobře, jako absorber nárazu při dopadu. 
Antény je však potřeba po nárazu do země vyměnit za nové, neboť typicky dojde k prasknutí plastových dílů. 
Překážkou k této operaci je nutnost odletování a zpětného zaletování smyček antény na přívodní koax, což je časově náročná operace, která trvá přes hodinu. 
Podstatně by to však mohly změnit dvě konstrukční úpravy.

- [ ] Antény přichytit na suchý zip zespodu gondoly
- [ ] Vyřešit [potíž s připojením koaxiálního kabelu](https://github.com/cernohorsky/QFH-Antenna-868MHz/issues/1).

kaklik, 2020-10-19:
Výrazně by tomuto mohla pomoci úprav abalunu od @cernohorsky  
Viz: https://github.com/cernohorsky/AntennaBalun/issues/1

kaklik, 2020-10-27:
### Připojení antény ke koaxiálnímu kabelu 

@cernohorsky  Navrhl úpravu svého balunu k dipolové anténě, která řeší problém s připojením kabelu k anténě. 
Je zde: https://github.com/ODZ-UJF-AV-CR/QFHBAL01

@ChroustJan Ten plošný spoj jde rozměrově trochu upravit. Myslíš že pak bude možné jej přidat k nějaké aktuální výrobě PCB? 

### Uchycení antény na gondolu 

Anténa momentálně má na spodní straně vytištěnou mřížku, která sloužila k připevnení ke gondole lepící páskou. Tento způsob připojení se celkem osvědčil, ale není moc flexibilní.

[image]

Navrhoval bych mřížku upravit tak, aby v rozích byly plné čtverečky na které se přilepí [samolepící suchý zip](https://www.azktrade.cz/3m-sj3550-dual-lock-samolepici-suchy-zip-cerny-4/).  Stejný suchý zip by pak byl nalepen ze spodní strany na gondole. 
Anténu by mohl upravit někdo, kdo dovede pracovat s OpenSCADem. Třeba @Alsa-Tucnak  ?

Alsa-Tucnak, 2020-11-03:
Anténu upravit můžu. Jen je potřeba zajistit opravdu dobře její uchycení. U té mřížky mi přijde, že je dost velká p-st, že se od spirály odlomí. 4 suché zipy v rozích by mohly držet dobře snad. Zkusím tedy sehnat suchý zip a upravit anténu.

kaklik, 2020-11-03:
> Anténu upravit můžu. Jen je potřeba zajistit opravdu dobře její uchycení. U té mřížky mi přijde, že je dost velká p-st, že se od spirály odlomí. 4 suché zipy v rozích by mohly držet dobře snad. Zkusím tedy sehnat suchý zip a upravit anténu.

Super! :)  Nevím jak to myslíš s tím odlomením mřížky od spirály.  Přijde mi, že by došlo k úplně destrukci antény, kdyby se tohle mělo ulomit. 

Jinak do toho komentáře výše jsem dal odkaz na konkrétní typ suchého zipu (dual-lock), který by se tam myslím měl použít.

kaklik, 2020-11-06:
Jinak prostřední část té 3D mřížky ve spodní části antény bude potřeba upravit tak, aby na ní bylo možné připevnit [tento plošný spoj](https://github.com/ODZ-UJF-AV-CR/QFHBAL01).

Dovedu si představit, že by to bylo realizované tak, že na mřížce budou vytištěny 3 trny, které zapadnou do třech montážních děr tohoto plošného spoje.  Tím by byla i definována jeho orientace, neboť tam musí být upevněn ve správné rotaci. 

Potom ty tři vytištěné trny půjdou rozmáčknout něčím horkým, aby z nich plošný spoj nespadl. 

Anténa by se tak sundavala jako celek, kdy se připojí kabel do konektoru plošného spoje a celá anténa se přilepí na suché zipy.

kaklik, 2020-11-22:
Zde jsem [exportoval rozměrový výkres QFHBAL01](https://github.com/ODZ-UJF-AV-CR/QFHBAL01#mechanical-drawing)

[image]

kaklik, 2020-11-24:
Nový kód jsem sloučil s původnim kódem, aby nevznikalo  příliš mnoho variant (původní anténa už nemá žádné využití). 

Řešení kdy plošný spoj bude nasazen na kuželové trny, je docela revoluční, ale asi dobré. Neboť takhle není nutné řešit přesné tolerance otvorů :)

[image]

Průměr otvoru v PCB je 3.2mm. To jsem na obrázku výše zapomněl. Pro aktuální trn to znamená, že zřejmě plošný spoj bude někde lehce pod polovinou výšky trnu. Což je zřejmě správně. :)

Potíž nastává s umístěním trnů a s otvorem pro koaxiální konektor, který musí být vyveden zřejmě ze spodní strany. 

Plošný spoj myslím si, bude umístěn osově přesně na druhou stranu. Tj. takhle. (Tohle pak musím ještě znovu zkontrolovat - v kódu by na to bylo dobré udělat přepínátko.)

[image]

Zatím jsem do antény doplnil pouze vrstvu "QFHBAL01-User_Comments.dxf" na které jsou vidět kóty a obrysy. 

Další potíš je domnívám se ta, že aktuálně pozice RF konektoru vychází přesně do křížení mřížky. 
Tady jsem vytvořil nový obrázek, který má doplněny kóty ke konetoru. 

[image]

Konektor potřebuje ze spodní strany kruh volného prostoru o průměru minimálně 8mm, aby do něj bylo možné připojit nebo odpojit koaxiální kabel.

kaklik, 2020-11-25:
Tak správná orientace QFHBAL na spodní straně antény je tahle: 

[image]

Čevené značky SCAD souboru jsou sesouhlaseny s orintací PCB na levém obrázku.

Parametry rozměrů, které jsou nyní [přímo v kódu antény](https://github.com/ODZ-UJF-AV-CR/QFH-Antenna/blob/e0976c53d5830cff4555565b1e6811471ef7c436/QFH-Antenna.scad#L47-L57) by měly být přesunuty do [souboru s parametry](https://github.com/ODZ-UJF-AV-CR/QFH-Antenna/blob/master/parameters.scad).

kaklik, 2020-12-06:
Antény byly přilepeny 3M dual-lock suchým zipem na víko gondoly.

closed 2020-12-06 by kaklik