ODZ-UJF-AV-CR/balon #42: V gondole používat různé napájecí zdroje

kaklik, 2019-12-02:
Při letu Fík5 se příliš neosvědčilo použití [primárních lithiových článků](https://www.tme.eu/cz/details/saft-ls33600/baterie/saft/ls-33600/). Díky jejich nízké proudové zatížitelnosti, která během krátkého letu tlakového stratosférického balonu v podstatě neumožňuje využít plnou kapacitu článku.   Důsledkem tak je, že část hmotnosti článků je v podstatě nevyužitelný balast. 

Lepším řešením by tak zřejmě bylo tyto primární lithiové články nechávat spíše pro referenční měření a záložní dohledávací systém, kde je předpoklad nízké spotřeby a potřeba dlouhé provozní výdrže. 

Pro hlavní experiment by tak bylo vhodnější použití [LION2CELL01D modulu](https://mlab.cz/module/LION2CELL01D) vybaveného li-Ion akumulátory pro střední proudy.  Z tohoto modulu je pak potřeba vytvořit 5V napájení (autopilot a telemetrie) a 3.3V napájení pro experimentální elektroniku.
Pro případ, že dva články v serii nemají dostatečnou energetickou kapacitu pro spolehlivé napájení experimentu na ~6h provozu, tak by bylo potřeba mít řešení pro paralelní zapojení další dvojice lithiových článků k již existující dvojici na modulu.

Sommema4, 2020-10-23:
Znamená to teda, že se tím teda nebude napájet ten iontometr, jak jsem navrhoval na schůzce?

kaklik, 2020-10-23:
> Znamená to teda, že se tím teda nebude napájet ten iontometr, jak jsem navrhoval na schůzce?

Já tam teda vidím napsáno "Pro hlavní experiment by tak bylo vhodnější použití LION2CELL01D modulu vybaveného li-Ion akumulátory pro střední proudy. "  Tvůj, iontometr je hlavní experiment, ne? :)

kaklik, 2020-12-06:
V gondole se používá kombinace napájení z primárních článků AIRDOSu a nabíjecích akumulátorů v modulu LION2CELL, který má napájecí vstup vyveden mimo krabici gondoly a bude připojen k napájení až do okamžiku startu.

closed 2020-12-06 by kaklik